blueman-project/blueman #2819: Automatic Diconnect

justpurv, 2025-08-02:
i don't know why i have tried all the solutions but blueman automatically disconnects my bluetooth device. i am sick of it

cschramm, 2025-08-02:
blueman does not do anything like automatically disconnecting devices. How did you come to that conclusion? Did you even try without blueman running?

rahulraikwar00, 2025-08-03:
yes , i am facing the same issue , now its not even connecting . i did update the drivers

justpurv, 2025-08-03:
Man I am not roasting the app. I am trying to find the solution here so
would you please show your love for the app somewhere else.

cschramm, 2025-08-03:
So there is no indication that blueman is actually triggering any disconnection? I'd guess that you're rather witnessing a failed connection attempt than. I'll convert this into a general Bluetooth discussion.