blueman-project/blueman #2828: blueman-applet missing "connect" in context menu for Stadia controller

Jinra, 2025-08-07:
For some reason, in the applet, when right clicking on my Google Stadia controller to open the context menu, the "connect" button is missing. In the screenshots below, you can see that it exists for my other devices such as my headphones.

This is problematic since the Stadia controller will not auto connect to my computer at first boot (or first startup of the bluetooth systemd service; turning the controller off and on does have it reconnect automatically).

If I right click on the system tray icon, I can hit "reconnect to stadia" and it will work fine. Additionally using `bluetoothctl connect <stadia_mac_address>` also works in terminal.

Missing connect button

[image]

Connect button exists on other devices

[image]

The system tray workaround works well enough for me, just thought I'd report this as it seems like an easy UI fix.

Jinra, 2025-08-07:
@cschramm Oh good to know! Is the tray icon "reconnect to.." option not doing the a similar connection call to the device?

* CachyOS (6.16.0-30-cachyos-lto)
* Bluez v5.83-1.2

FWIW in KDE the behavior is similar, though the bluetooth applet in there (bluedevil I think?) had a connect button to let me easily connect on first boot

infirit, 2025-08-08:
> However, generally speaking, when trying to connect to a device that does not advertise any such services, there is a high chance of just causing failures. I'm thus not a fan of offering the connect action unconditionally.

Yeah, I don´t think there is a good solution for this, but.. How about we change the menu item to have a different icon and an updated tooltip? But to be clear, only for devices that are Paired already and do **not** provide one of the generic connect profiles.

[image]

@Jinra can you show a screenshot of the Info window when blueman-manager does __not__ allow you to connect the stadia controller. Just want to make sure the relevant UUIDs are in fact missing for you.

Jinra, 2025-08-08:
@infirit here you go

[image]

closed 2025-10-15 by github-actions[bot]