blueman-project/blueman #2829: Blueman-Manager Buttons Unresponsive and Paired Devices Disappeared

revo-cs, 2025-08-07:
The buttons in blueman-manager (e.g., Add, Connect) are unresponsive, and all previously paired Bluetooth devices disappeared overnight. 

The Bluetooth service is enabled, but the manager fails to detect devices or allow interaction. This issue started suddenly after the application worked correctly the previous day.

My Environment:
1) Operating System: Cinnamon v: 6.4.8 Distro: Linux Mint 22.1 Xia
2) Kernel Version: 6.8.0-71-generic
3) Blueman Version: 2.4.4+mint2+xia
4) Bluez Version: 5.72-0ubuntu5.3
5) Hardware : Bluetooth 
  Device-1: Intel AX200 Bluetooth driver: btusb v: 0.8 type: USB
    bus-ID: 1-10:5
  Report: hciconfig ID: hci0 rfk-id: 0 state: down
    bt-service: enabled,running rfk-block: hardware: no software: no
    address: <filter>          
6) Bluetooth Adapter: Bus 001 Device 005: ID 8087:0029 Intel Corp. AX200 Bluetooth

Results:

dmesg | grep -i bluetooth
[    2.691561] Bluetooth: Core ver 2.22
[    2.691616] NET: Registered PF_BLUETOOTH protocol family
[    2.691619] Bluetooth: HCI device and connection manager initialized
[    2.691627] Bluetooth: HCI socket layer initialized
[    2.691630] Bluetooth: L2CAP socket layer initialized
[    2.691636] Bluetooth: SCO socket layer initialized
[    3.590250] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    3.590254] Bluetooth: BNEP filters: protocol multicast
[    3.590258] Bluetooth: BNEP socket layer initialized
[    4.942912] Bluetooth: hci0: Reading Intel version command failed (-110)
[    4.942918] Bluetooth: hci0: command 0xfc05 tx timeout

rfkill list
0: hci0: Bluetooth
	Soft blocked: no
	Hard blocked: no
1: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: no

bluetoothctl
Waiting to connect to bluetoothd...[bluetooth]# Agent registered

pactl info | grep Server
Server String: /run/user/1000/pulse/native
Server Protocol Version: 35
Server Name: PulseAudio (on PipeWire 1.0.5)
Server Version: 15.0.0

alldispatchingwithus-lang, 2025-08-07:
hello i have the same problems any help please

ArielMedina, 2025-08-08:
I have the same problem, has anyone found the solution?