blueman-project/blueman #2839: bluetooth issue

tkboss6139-cmd, 2025-08-15:
I have mint when i connect Bluetooth to soundbar is connect but not working still sound come from laptop

kindly resolve the issue

warm regards

Tyyb

Alef-0, 2025-08-15:
I came here because my bluetooth wasn't working as well, I'm on linux Mint 22.1 in a laptop.
I've tried some stuffs then rebooted and now it's working, it began working while I was looking at this github tho.
So if you are trying to fix something, or someone is trying to find the problem what I did was:
``` bash
sudo apt purge bluez
sudo apt install bluez
sudo apt install blueman
```
First I uninstalled everything I could following a linux mint forum advice, this didn't work so I asked for a chat AI how to reinstall every bluetooth dependency.
``` bash
sudo apt-get autoremove
sudo apt-get install  --reinstall bluetooth pulseaudio-module-bluetooth blueman bluez-firmware bluez-tools
```
Now for some reason there was some error of broken packages, and I solved by installing the specific version it was required trough: `sudo apt install pulseaudio=1:16.1+dfsg1-2ubuntu10.1 pulseaudio-module-bluetooth`
Finally I rebooted and it began working.

Alef-0, 2025-08-15:
A little update, It _was_ working.
After I connected the bluetooth for the first time it worked fine then it suddenly disconnected, I needed to remove and reboot my computer again for it to connect, and then it happened again some time later.
Something is still wrong.