blueman-project/blueman #2860: Built-in Motherboard Bluetooth Disabled

peavysebastian, 2025-08-25:
Mint 22.1

Recently I had to restart my computer to resolve a seperate issue, and when I turned it back on my Bluetooth Devices applet was no longer detecting that I had an adapter, but rather the Adapter dropdown menu showed a series of empty options. I saw my Update Manager had a bluetooth update waiting and so I installed the update and restarted my computer, expecting it to work. No dice. My bluetooth was working with no issues yesterday, so I cannot imagine what the issue might be. For reference, I have a built-in bluetooth adapter on my motherboard, which is an X870 Steel Legend WiFi. I have not kicked or hit my tower in the intervening time. Does anyone have a fix?

A screenshot of my Bluetooth Device Manager, for reference:

[image]

MrBlue281, 2025-08-26:
I'm having the exact same issue. It seems to have happened with the update from version 5.72-0ubuntu5.3 to 5.72-0ubuntu5.4.

alexandermilona, 2025-08-27:
This is a temporary fix, but it has to be run every time the PC is turned on.

sudo modprobe -r btusb mt7925e      
sudo modprobe mt7925e
sudo modprobe btusb