blueman-project/blueman #2863: Mouse sleep/ standby ?

EdmundDuckDuck, 2025-08-30:
My logitech mouse goes to sleep mode if not used for more then a couple of seconds.
I have  to move or click the mouse to get is working which is very annoying.
My logitech keyboard works fine so it isn't a USB thing.

Linux mint 22.1 XFCE
Blueman manager 2.4.3

BTW received signal strength 100 % (too much )
Transmit power level 100 % ( very high )

cschramm, 2025-08-30:
Sounds like the intended behavior of the device, no?

EdmundDuckDuck, 2025-08-30:
I cannot find such a setting on the mouse and the mouse behaves correctly on my TV.

infirit, 2025-08-30:
This is mostly out of scope for blueman as it's mostly a frontend for the bluetooth service (BlueZ). Usually it's best to ask these sorts of questions on the forum of your distro.

My MX Master 3s does disconnect at some point but it takes minutes not seconds and it reconnects almost instantly. It may still be USB related as many Bluetooth adapters are essentially USB devices (even onboard ones). There is USB power saving which might cause the behaviour you're describe, see this for example https://askubuntu.com/a/1362310