blueman-project/blueman #2866: Disabling ConnectionNotifier does not stop notifications

Ao1Pointblank, 2025-09-01:
Blueman 2.4.6
EndeavourOS CInnamon

As the title suggests, I am plagued with connection successful notifications which I cannot disable by unchecking the Connections Notifier plugin. 

If there is a way to manually break this plugin, I would accept that as a solution.

cschramm, 2025-09-01:
Disabling the plugin should indeed have the desired effect. What exact notification are you still getting?

Ao1Pointblank, 2025-09-01:
[image]

Ao1Pointblank, 2025-09-01:
Aha, thanks. I have to disable audio and input profiles checkbox? 
What exactly does that option do, anyway?

jstolarek, 2025-09-30:
I have the exact same problem. I am running version 2.4.4 on Debian Trixie and despite having notification plugin turned off I still get notifications when connecting my wireless headphones:

[image]

How do I disable that? Disabling automatic connection of "Audio and input profiles" is not a solution, because it breaks headphone functionality.

Bezowski, 2025-11-26:
> I found a workaround: edit `AutoConnect.py` applet file, under Debian located in `/usr/lib/python3/dist-packages/blueman/plugins/applet`. Find the function responsible for the notification (line 65 in version 2.4.4):

> and comment out the three lines that display the notification:
> 
> def reply(dev: Optional[Device] = device, service_name: str = ServiceUUID(uuid).name) -> None:
>     assert isinstance(dev, Device)  # https://github.com/python/mypy/issues/2608
>     # Notification(_("Connected"), _("Automatically connected to %(service)s on %(device)s") %
>     #              {"service": service_name, "device": dev.display_name},
>     #              icon_name=dev["Icon"]).show()

Thanks, this worked for me on Linux Mint.

jstolarek, 2026-03-28:
Still not resolved.